alduxvm/pyMultiWii #2: Latest Cleanflight version 1.12

Arise, 2016-03-29:
Hi!

I can see in pyMultiWii.py that line:

>  Modification required on Multiwii firmware to Protocol.cpp in evaluateCommand:

Is the modification still necessary? The cleanflight code got changed over time and now that evaluateCommand is no longer there.

I'm trying the test-arm-disarm.py and nothing happens.

Also this part seems not updated. Cleanflight has rx_serial instead of serialrx.
and serialrx_type seems to be replaced with serialrx_provider?

>  but by default you will not be able to use the MSP_SET_RAW_RC to write pilot commands to the multiwii. In order to do that you need to activate (via the baseflight/cleanflight GUI) the SerialRX with the specific type for MSP (MultiWii Serial Protocol). The instructions for doing that on baseflight are:
> 
> Open the CLI (while on the baseflight configurator) and type:
> feature SERIALRX
> and then type the following lines:
> 
> set serialrx_type=4

alduxvm, 2016-03-29:
Thats an old message for 8bit flight controllers, is not even needed for those controllers actually, I crafted that message in order to receive a bigger message that I needed. You can ignore it. 

I see that you're using a naze32 or similar. I have not use cleanflight with my library... I use baseflight, and in baseflight you just need to activate "set serialrx_type=4".

For cleanflight... as I told you before, I have not tested (I always use baseflight), but it should have something similar. If you cannot make it work, then send another issue and I'll check it using cleanflight.

alduxvm, 2016-03-31:
Did you manage to make it work properly with cleanflight??

Arise, 2016-03-31:
Hello!

I just went closer and closer to the final goal.

I'll try to describe the steps.

In cleanflight it seems that it is possible to do multiple serial connections.
First UART1 is usually reserved to be able to log in via Cleanflight configurator, but is possible to connect over any other UART serial interface (Cables with Prolific chips or CH340/CH341A or maybe FTDI???) also.

There is UART1 (normal micro USB), UART 2 (pin 3&4) and softserial1 (pin 6&5) and softserial2 (pin 8&7).

So basically it is possible to make at least one extra link to the NAZE32 board, one over micro USB cable and another one serial connection on UART2 for example.

You can use UART2 to send MSP messages to the flight controller that is at the same time connected on UART1 over the USB cable. so is like one cable can do input and another can do output.

Haven't succeeded yet to open 3 or 4 serials to the controller, it seems it allows only a max of 2 connections right now. I've asked in the cleanflight bugtracker maybe they can enable a third connection.

Next, I was able to run/disarm script but it took me a while to do that and to understand how it really works, also I had some issues with motors not spinning first time, I had to make some adjustments.

Also, I was able to connect via EZ-GUI software first on the serial connection over an USB OTG cable.

But today I went a lil bit further and I've installed a raspbian on my rPI2, and added some apps via apt-get, like xrdc to remote into rpi, a custom chromium then I went and managed to install cleanflight configurator on raspbian, then did some more apps and also... ser2net which looks awesome.

ser2net lets you estabilish a connection to the serial port over network, so I went further and connected from EZ-GUI running on my android phone over a wifi connection to the raspberry pi board running the ser3net that forwards data to the ttyUSB0 that is the USB cable from rpi2 to the naze32.

I was able to arm and see the parameters.

Hopefully tomorrow I will be able to finish connecting everything and make everything autonomous, then I plan to make it fly or ... something. :)

yconst, 2016-07-17:
Hi Arise,

I am trying to achieve the same using ser2net. Can you please post the settings you used for /etc/ser2net_conf?

Thanks

Arise, 2016-07-19:
Sorry, I don't remember and not sure if I still have the image to take a look.
But the protocol worked easily after a few attempts.

benne333, 2016-09-02:
Hi,
I've encountered the same issue, but with Cleanflight (1.13) AND Baseflight also. Actually I hoped to have a very simple start without messing with additional cables - just connecting the PI with USB and start off :-). From looking in the code, I suspect that channel configuration maybe the reason? In sendCMD you only send 4 values, my configuration says 8 (include AUX). 
Any hints?
Regards,
Bernd

alduxvm, 2016-09-03:
The best solution to make it work, is to use Baseflight. I designed the library using baseflight (it was the only one back then...) and then cleanflight was created and gain more popularity... But, unfortunately I did it to work on baseflight... 

The RC commands on MSP are defined as: `ROLL/PITCH/YAW/THROTTLE/AUX1/AUX2/AUX3AUX4` 

You can send either just 4 or the entire 8...

I do have one question... why do you need to use cleanflight? what does it have that baseflight does not? that is essential for your work with pymultiwii?

benne333, 2016-09-03:
Hi,

> The best solution to make it work, is to use Baseflight. I designed the library using baseflight (it was the only one back then...) and then cleanflight was created and gain more popularity... But, unfortunately I did it to work on baseflight...
I gave baseflight a (probably too) short test. It didn’t work there either. But I assume that I have to deal more with some configuration values (valid ranges of input or similar). Any hints there? I’m new to FC’s and do this as a project with my 14y old son. My interest is to bring flying as quick as possible to higher level programming, because thats what I’m familiar with :-). Looking in using OpenCV or others ...
> 
> The RC commands on MSP are defined as: ROLL/PITCH/YAW/THROTTLE/AUX1/AUX2/AUX3AUX4
> 
> You can send either just 4 or the entire 8...
OK, so this shouldn’t be the problem.
> 
> I do have one question... why do you need to use cleanflight? what does it have that baseflight does not? that is essential for your work with pymultiwii?
The reason is - as you wrote above - it seams to be more popular nowadays... and the development of baseflight seems to be dead.
Thanks so far, any hints always appreciated :-),
Benne

conradv5379, 2017-03-12:
Hi there!  I am trying to get my Naze32 with Baseflight to arm or to send a command but to no success.  I have tried CleanFlight and also iNav already before but since Baseflight is recommended, now I am trying to stick with Baseflight for testing purposes.  Can anyone please direct me to a solution please?  I have also tried to use UART2 but still the same issue.  I can read data but not send commands from pyMultiWii.

Thanks in advance

alduxvm, 2017-03-12:
When using baseflight, did you activated ```feature SERIALRX```and ```set serialrx_type=4``` in the GUI of baseflight?

conradv5379, 2017-03-13:
Yes I did and verified from the CLI.

alduxvm, 2017-03-13:
can you send me an email to aldo@altax.net so I can give you further guidance?

Anderson321, 2017-04-27:
Was this issue ever figured out? I am having the same issue with baseflight and cleanflight. I am unable to send commands. @conradv5379 @alduxvm Thanks!

benne333, 2017-04-28:
Not really. What helped in my case was to configure arming on AUX1. However, later I was able to arm by sending the usual stick commands, but not reliable, when I remember correctly. Maybe worth a try!

Regards,
Bernd Dreier

Anderson321, 2017-05-11:
@Arise Hi since you are one of the people to figure this out, how do you send commands? I'm using cleanflight, a naze32 and i have RX MSP enabled. I am able to get attitude but I can't send Commands. How did you do it?

alduxvm, 2017-05-11:
Apparently the CLI process to activate the serialRX has changed in the new baseflight release, therefore that part needs to be figure it out with the developers of baseflight, you can try an older version or baseflight and set it up. At the moment I don't have a naze32 board to be able to test.

Anderson321, 2017-05-12:
[image]

My configurator looks like this and there is an option to enable RX MSP. Aren't your CLI functions doing the same thing as this?
"feature SERIALRX
and then type the following lines:
set serialrx_type=4"
Even if you can't do it through the command line, I feel like selecting RX MSP in the configurator would do the same thing.

alduxvm, 2017-05-12:
In the screenshot, RX_SERIAL is selected, can you try and select RX_MSP and then test the arm-disarm script? if it does not work, the please ask for a method to activate MSP in the baseflight github, or downgrade your bf version.

Anderson321, 2017-05-12:
Sorry, I dont have my naze32 with me right now so I just googled a picture of it and took a screenshot. My configurator just looks like this but RX_MSP is selected. What version should i downgrade too?

conradv5379, 2017-05-12:
I have successfully managed to control cleanflight using msp.  The catch is in the order of the channels (Pitch, Roll, Throttle,Yaw etc....)  when you send the Set_RAW_RC command.  To test, I suggest that you have the Cleanflight configurator open and select the Receiver tab so that you can analyse the channel values and order to match with your code.

alduxvm, 2017-05-12:
Then that means that the channel order in cleanflight is different than in baseflight. 

In the original MSP (http://www.multiwii.com/wiki/index.php?title=Multiwii_Serial_Protocol), the `MSP_SET_RAW_RC` uses this order: `ROLL/PITCH/YAW/THROTTLE/AUX1/AUX2/AUX3AUX4`.  

But with @conradv5379 comments, it means the cleanflight devs change it to: `ROLL/PITCH/THROTTLE/YAW/AUX1/AUX2/AUX3AUX4`, interesting... 

Thanks everyone!

Anderson321, 2017-05-13:
@conradv5379 When I use this python code to connect via serial, cleanflight disconnects from the flight controller so I can't see the channels being updated in the receiver tab. Also, why would the order of the channels matter? When i execute the code it gets to "b = self.ser.write(struct.pack('<3c2B%dHB' % len(data), *total_data))" and then nothing happens

conradv5379, 2017-05-13:
Connect a Bluetooth module to one of the UARTS and set MSP on that UART. You should be able to connect to clean flight configurator through Bluetooth while sending MSP commands through the usb port.

Anderson321, 2017-05-13:
@conradv5379 I'm a little confused on how the channel ordering matters? Inside the arm function it sends a command to self using self.sendCMD(8,MultiWii.SET_RAW_RC,data) where data is data = [1500,1500,2000,1000].  And then in sendCMD function it does b = self.ser.write(struct.pack('<3c2B%dHB' % len(data), *total_data)) . But then nothing happens? I don't even know how to debug this

Anderson321, 2017-05-14:
Also I tried to print struct.pack('<3c2B%dHB' % len(data), *total_data) and all i got was "$M??????" in the terminal. If that's being written to the serial, that doesn't seem right.

Arise, 2017-05-14:
Hello guys!
I've give up using various devices to control the quad copter because it just won't work safely. At that time I had only the quad, with no remote, so I was trying to control it via my phone. After the first crash, which occurred pretty fast after I raised it in the air (which was sheer luck, because it could have been a run away) I decided is not worthy to try to control it that way.

So, if you were in that situation, then please do yourself a favor and buy a proper remote transmitter.

Otherwise, I would love to see other scenarios when you are using MSP commands.

Please note than Cleanflight 2 is actually a Betaflight 3 rename, some MSP commands no longer works, I'm not sure about this but I've seen recently a bug report/note about some MSP not working properly with Cleanflight 2 but only with Cleanflight 1.

Anderson321, 2017-05-15:
@Arise Sorry to bother you but which version of cleanflight worked for you? Also did you change @alduxvm's python code at all? I downgraded to 1.14 but I still can't send commands

conradv5379, 2017-05-17:
I had to change the arm and disarm code methods in pyMulitiwii file. It works with the latest CleanFlight 2.

Arise, 2017-05-17:
At that time I was on Cleanflight 1.x, it wasnt merged with Betaflight at that time.

snndk, 2017-05-27:
@conradv5379 Hi! These are my settings
[image]
[image]

And im sending set_raw_rc command with your order(pitch,roll,throttle,yaw). What am i missing ? 
BTW my CF conf. VERSION is 2.0.3

HardVeur, 2018-05-13:
Did someone get it work with cleanflight?! I actually doesn't care if I use Cleanflight, Baseflight or Betaflight the only problem is I can't find any Binaries for Baseflight since baseflight.net doesn't exists anymore and either Cleanflight or Betaflight support the 'set serialrx_type=4' command in CLI anymore so I can't get any work around for MSP_SET_RAW_RC. Did anyone made it?!