alduxvm/pyMultiWii #3: Some issue with python 2.7 and 3.. like print have no parenthesis and exception have comma instead of as

waqarrashid33, 2016-06-11:
I will try to fork it and fix it as i get time..but right now I had to complete my drone project

alduxvm, 2016-06-11:
can you explain more about your error? perhaps paste some output of the terminal?

waqarrashid33, 2016-06-12:
Right now I don't have the output, I tried to fix some of the errors but now the data receiving doesn't seem consistent..there are a lot of "None" prints.. I will post the error as soon as possible.. I am working on a student project on a quadcopter and the deadline is a bit close..

waqarrashid33, 2016-06-12:
some issues that I remember are missing parenthesis in print.. like you used `print "Hello" `instead of `print("Hello") `and instead of `except Exception as error:` you used `except Exception, error:`..

alduxvm, 2016-06-12:
Alright, this library was made to work on p2.7, there is no support for p3, that might be your main issue. But if you don't show me output and what where you doing, then is very difficult for me to guess your issue. Please send me your configuration, OS, board, what script you use, and output.

There is sometimes `None`, when there is no info coming back from the board... In an email you sent me, you told me you had a "multiwii 2.4" board, thats an old one, with an 8bit processor. At the moment I use this library a lot on naze32's, this ones are much much faster... to make the library work with old boards like yours, you need to change the `wakeup = 2` on line 84, to: `wakeup = 14`, the old boards need more time to "boot up".

closed 2016-12-12 by alduxvm