alduxvm/pyMultiWii #4: Problems when trying to control via MSP

enriquemc96, 2016-06-14:
Hi

I've been running the tests and I only was able to run correctly the test-attitude.py. I enabled the feature SERIALRX and set SERIALRX_TYPE to 4 with CLI in BaseFlight but I'm not able to write pilot commands to the board and I don't know if I'm doing something wrong.

I have a naze32 with the last stable BaseFlight firmware.

Thanks

alduxvm, 2016-06-14:
Of course I follow this thread...

Please show me the output when you try `test-arm-disarm.py`...

did you solved your issue?

cheers

enriquemc96, 2016-06-14:
Hello

The problem is solved, it was an absolutely silly mistake... I didin't know that there is a maximum angle permited for arming and I had the board in a position with a higher angle than it, so... But thanks for answering.

closed 2016-06-14 by alduxvm