alduxvm/pyMultiWii #5: Versioning + Packaging and uploading to PyPI

amorphic, 2016-06-22:
Thanks for an excellent Python module. We're experimenting with wifi drone flying at our makerspace [SparkCC](http://sparkcc.org/) and pyMultiWii saved us a lot of work!

It would be great if pyMultiWii was properly versioned and made available on [PyPI](https://pypi.python.org/pypi) so that it could be easily installed via [pip](https://pip.pypa.io/en/stable/) and included as a dependency in other projects.

Happy to help out with this if you'd like me to.

alduxvm, 2016-12-12:
Hi! 

This sounds great!! (so sorry for not replying earlier...), yes I will be interested in doing that, how should we start doing it?

cheers!

probhakarroy, 2017-09-23:
Just checkout this site!!!
http://peterdowns.com/posts/first-time-with-pypi.html

probhakarroy, 2017-09-24:
Just checkout this site!!!
http://peterdowns.com/posts/first-time-with-pypi.html

And thanks for such a great module!!

closed 2019-04-02 by alduxvm