alduxvm/pyMultiWii #6: I have a problem with cc3d with cleanflight and I goy two of them another i9s atom

kopterimies, 2016-09-12:
channels 5-6 dont work? motors 5 and 6 dont spin.

alduxvm, 2016-09-14:
Hi, can you try and use baseflight? that's the firmware I use... I haven't use cleanflight and I explain in the instructions that this one supports only baseflight... 

Now, have you tried test-arm-disarm? does that work? If that works, then you shouldn't have issue... can I see the snippet of code you're using? that will help.

kopterimies, 2016-09-15:
sorry I`dont yet understand this github and code system how to work?

closed 2016-12-12 by alduxvm

kopterimies, 2017-01-26:
Now my number one problem is how to flash open pilot revolution back to
what it was. It has much much more than cleanflight or beta or baseflight.
It was big mistake to flash it to cleanflight. New cleanflight without
problems was 32 € and revolution has automatic mode with GPS and everything
else. So how to flash it back.