alduxvm/pyMultiWii #8: feature SERIALRX Invalid name

bigboateng, 2017-03-07:
I want to use the library to send commands to my naze32,  but typing `feature SERIALRX` in the CLI gives invalid name error. Is this due to the version of cleanflight? If so do you know what the new command is, Many thanks.

alduxvm, 2017-03-07:
Hi, yes, cleanflight is not supported, only baseflight is used and tested with this library. Probably in the future I will add support for cleanflight. Cheers!

bigboateng, 2017-03-07:
Thanks for reply, is the firmware uploaded through baseflight different from the one uploaded through cleanflight? because I just tried same command and baseflight says invalid name for `feature SERIALRX`.

alduxvm, 2017-03-07:
yes, the firmwares are different... Please reflash your board with baseflight and follow the instructions for activating the SERIALRX commands.

bigboateng, 2017-03-07:
It worked now, thanks!

closed 2017-03-07 by bigboateng