alduxvm/pyMultiWii #19: Error with sendCMD

HardVeur, 2017-12-28:
It seems like the ord() get a int instead of a character that’s what cause the error in my case any suggestions to fix this problem?

nik012003, 2017-12-28:
Are you using python2.6 ?

HardVeur, 2017-12-28:
Wait did you use python 3.6 cause I did and I couldn’t see any methods which probably behave different from the 2.6 version. So I thought it should also work with 3.6 but lessons learned i didn’t.
Do you mean the version of multiwiiMSP you posted in your issue???

HardVeur, 2017-12-28:
I use the CF 2.2.0 wich i flashed on my CC3D but someone from the CF-Forum told me that 2.2.0 CF doesn’t support serial rx (MSP) anymore. Could you explain how do you got your whole set up run and how does your setup look like? I really would appreciate that, thank you.

nik012003, 2017-12-29:
I haven't test serialrx yet, but the option is there on CF 2.2.0 on the SeriuslyPro F3.
I'm able to get data(ATTITUDE,ALTITUDE,etc...) from my FC via python 2.6 (the lib is incompatible with 3.6).
The only thing I got enabled on the CF configurator is MSP on the USB(UART1 in my case).
When I'm not busy I'll try to get serialrx to work, but I think it needs a separate UART connection.
I'll keep you updated.

alduxvm, 2017-12-29:
Yes, I created the library with support only for 2.7, if someone needs to support 3, then it will have to be re-done...

HardVeur, 2017-12-29:
Thank you guys for the help. I try hard to get my setup run. I have a CC3D and try to get a list with 4 integers on my raspberry pi to my CC3D with Cleanflight version 2.1.0 (I said 2.2.0 before that was a mistake it’s 2.1.0). How to get the list with throttle, yaw, roll and pitch from my raspberry to my CC3D doesn’t matter as long as I get it done and I’m ready to flash any firmware to my CC3D as long as I get this run. Any suggestions how to do it? I already tried to use the MuliWii python class with sendCMD(8, MultiWii.SET_RAW_RC, [throttle, yaw, pitch, roll]) but it didn’t worked I also tried sendCMDreceiveATT and it didn’t worked either I just get none as respond.
How to get the list from raspberry pi to CC3D should I flash a specific version of Betaflight to my CC3D?

nik012003, 2017-12-29:
I also got None in my tests, I fixed the problem with this PR #18 .
Try to use my version and see if it fixes your issue.

HardVeur, 2017-12-29:
Actually I just tried your version and never the original one. But your changes just give the board a couple of seconds to start. So I thought it can’t be wrong to just test yours or do they also something else I don’t remember the exact changes right now?!

nik012003, 2017-12-30:
My version changes a bit the way the board connects and works perfectly in my setup.
I have got only one FC to test with so I don't know if it works on other FCs.
Maybe you could give alduxvm's version a try and see if that works for you.
BTW I would suggest that you change your CC3D to a newer board, with more powerful processors, like the F3,F4 or the F7.

engineer-186f, 2018-06-24:
I'm aware that I made this comment half a year later.

If anyone is still interested with controlling your MultiWii based drone, 
I can gladly inform everyone that I've made a module for Python 3.

@alduxvm You have my thanks for your hard work and effort.

nik012003, 2018-06-25:
Thank you very much @engineer-186f !
Have you tested it with betaflight/CF/iNav?
I'll try it on my quad as soon as possible!

engineer-186f, 2018-06-26:
@nik012003 I've only tested it without any systems like CF or Betaflight and it works just fine.
I can test it with these systems and update and later update the description of my script.

nik012003, 2018-06-28:
@engineer-186f it seems to get all the data from Betaflight-based systems too! 
But I haven't tested sending data yet.
[image]

engineer-186f, 2018-07-02:
@nik012003 That's great. I'm planning on implementing almost all necessary functions and operations for the board copter itself, including WP navigation and much more.

engineer-186f, 2019-02-04:
@LevanSanadiradze

engineer-186f, 2019-02-04:
@LevanSanadiradze 

Can you elaborate further? 

Multiple users have reported my module to be fully functional, even with CF/BF. 

Try to configure your firmware settings...

closed 2019-04-02 by alduxvm