alduxvm/pyMultiWii #20: pymultiwii library issue with inav 1.6

amitmanish, 2018-01-12:
Hello alduxvm, I tried using your pymultiwii library for controlling naze32 rev6 through raspberry pi. I can read board data like accelerometer value but not write commands like arm and disarm. I configured naze32     
with rc type to msp serial rx. raspberry pi and naze32 are connected through usb cable.

alduxvm, 2018-01-16:
most the issues trying to send cmd to the board via msp are related to firmware rather than this code... which firmware are you running?

amitmanish, 2018-01-17:
I am using inav version 1.6 and i have configured receiver type to serial_msp.

amitmanish, 2018-01-20:
Hello alduxvm, i checked the msp code of the inav but could not figure the reason for this inconsistency. Could you direct me where the possible inconsistencies can be which i need to change.

alduxvm, 2018-01-22:
Hi, I'm sorry, I created this library using baseflight... this was done a while ago, and now there is tons of different new firmwares that use the MSP protocol, without proper testing and debugging, I cannot make sure it will work with a no-supported firmware. That being said, I remember someone a while ago telling me they made it work with inav... can you please try to put baseflight to the naze32 and then try the arm-disarm and if it works, it means inav needs to be debugged...

amitmanish, 2018-01-23:
Thanks alduxvm. I will try this with baseflight.

closed 2018-01-23 by amitmanish