alduxvm/pyMultiWii #22: SPracing F3 V1 flight controller

jkurohara, 2018-08-05:
I am trying to read raw IMU data from a SPracing F3 V1 flight controller, but I can't seem to get any data.  Looping the getData function returns "None".  Any advice?  Thanks.

alduxvm, 2018-08-08:
ok, did you tried ATT first? does that one works?

jkurohara, 2018-08-10:
What is ATT?  Thank you!

alduxvm, 2018-08-10:
Attitude

nicolas-coulon, 2018-08-14:
I have the same issue.
```python
from pymultiwii import MultiWii
import time

serialPort = "/dev/ttyACM0"
board = MultiWii(serialPort)
while True:
    board.getData(MultiWii.ATTITUDE)
    print(board.attitude)
    time.sleep(0.1)
```
Attitude works with Betaflight
Configurator: 10.4.0
Firmware: BTFL 3.2.0 (Target: OMNI)

nicolas-coulon, 2018-08-14:
For me it's working now, I used python 3.6.5. It's working with python 2.7

closed 2019-04-02 by alduxvm