alduxvm/pyMultiWii #28: MSP V2 protocol support

wx4cb, 2019-03-05:
I know this is an "old" project, but is there any way that you can update to use the MSP V2 protocol?

it is described here: https://github.com/iNavFlight/inav/wiki/MSP-V2

Yes, i'm trying to talk to an INAV flight controller and it appears that it's because they use V2 of the MSP protocol.

nik012003, 2019-03-06:
You can use OpenGS. It's on my GitHub and it's specifically made for iNav.

wx4cb, 2019-03-06:
i don't see an open GS repo in your profile or am i looking at the wrong place?

https://github.com/alduxvm

nik012003, 2019-03-06:
I'm nik012003 not alduxvm.
Here's the link
https://github.com/nik012003/openGS

wx4cb, 2019-03-06:
@nik012003 oops my bad. Thanks for the link. I'll take a proper look when i get home.

nik012003, 2019-03-06:
@wx4cb No problem. If you want any help feel free to contact me on Skype (the username is nik012003).

closed 2019-04-02 by alduxvm