alduxvm/pyMultiWii #32: sendCMD() missing 1 required positional argument: 'data_format'

krish-iyer, 2020-01-17:
While running test-arm-disarm.py, it shows following error.
```
Error on Main: sendCMD() missing 1 required positional argument: 'data_format'
```

alduxvm, 2020-01-17:
which python are you using?

krish-iyer, 2020-01-18:
Python 3 but I am not very sure if it's version dependant. While sending a command, the data has to be packed for ROLL/PITCH/YAW/THROTTLE/AUX1/2/3/4. So I think it's necessary to specify data_format.

krish-iyer, 2020-01-21:
Something like this would work. It's not at all the best solution but exposes the trick perfectly. I have also tested out the code.

https://github.com/krish-iyer/pyMultiWii/commit/7ac3e8f237c5f4b0f528bb7633c88ae20d11b823

mgiglia92, 2021-11-23:
Any updates on this? I am also getting the same error. Planning on using this library for a drone competition project at university (im the technician). 
@krish-iyer, i've tried your fix, but it's giving me another error similar to when I tried fixing it on my own. It's a packing error. struct.pack is expecting 18 items for packing (got 26).

Hoping this is fixable....Are there any other implementations of MSP for python or will I have to write my own for the students for robustness?

alduxvm, 2021-11-24:
this means something change in the receiving part, before only 4 inputs where needed, not apparently 8 are needed, then what you need to do is modify the arm() function and add those 4 extra inputs needed and test.

But before this works, also SERIAL_RX needs to be enabled and you need to test that the board is receiving the commands and thats a big issue.

BTW which firmware are you running?

Niko-La, 2023-09-18:
> Something like this would work. It's not at all the best solution but exposes the trick perfectly. I have also tested out the code.
> 
> [krish-iyer@bac4a02](https://github.com/krish-iyer/pyMultiWii/commit/bac4a02cb611bdd0992b0e6e3de4bdcd27677f95)

@krish-iyer did this fix the tuple issue? and able to arm and sendcommands?

mgiglia92, 2023-09-18:
> > Something like this would work. It's not at all the best solution but exposes the trick perfectly. I have also tested out the code.
> > 
> > [krish-iyer@bac4a02](https://github.com/krish-iyer/pyMultiWii/commit/bac4a02cb611bdd0992b0e6e3de4bdcd27677f95)
> 
> @krish-iyer @mgiglia92 did this fix the tuple issue? and able to arm and sendcommands?

I unfortunately never got this working I ended up going a different route. I don't remember at this moment though

Niko-La, 2023-09-24:
> @Niko-La So as I recollect, the library worked fine. I think issue was with the python version. Did you check your python version? I am no longer working with the library or drones but definitely this library worked well.
> 
> Later I just created my own library. Here's the [documentation](https://hackdrones.readthedocs.io/en/latest/) and [code repository](https://github.com/krish-iyer/HackDrones/). It all boils down to how you unpack and pack packets.
> 
> Good luck!

@krish-iyer 
I'm aiming to control my device directly through MSP via USB VCP, using a Pi Zero W, without the need for any radio receiver.

I used your lib its nice clean up of @alduxvm original 
My Python script on the Pi is working efficiently to retrieve both altitude and attitude data. However, I've been encountering challenges with arming, disarming, and sending raw MSP RC commands through the USB port.

It's worth mentioning that I had this setup working seamlessly back in 2016. Since then, many updates and changes have occurred, so I'm in the process of revisiting and adjusting my approach.

I dont think `feature SERIALRX` and `set serialrx_type=4` are needed no more as by default msp is enabled on usb vcp.

To be clear you were getting it o arm, disarm, and pilot the motors using the usb-c port or you had to wire in a 2nd usb through another uart for pilot commands?

My code for pilot control

```
    board.arm()
    print ("Board is armed now!")
    print ("In 3 seconds it will disarm...")
    time.sleep(3)
    board.disarm()
    print ("Disarmed.")
    time.sleep(3)
```

```
    def arm(self):
        self.send_cmd(self.MSP["SET_RAW_RC"],self.safe_arm)
        print(self.MSP["SET_RAW_RC"],self.safe_arm)
        time.sleep(0.25)
        self.send_cmd(self.MSP["SET_RAW_RC"],self.arm_data)
        print(self.MSP["SET_RAW_RC"],self.arm_data)

    
    def disarm(self):
        self.send_cmd(self.MSP["SET_RAW_RC"],self.disarm_data)

    def alt_hold(self):
        self.send_cmd(self.MSP["SET_RAW_RC"],self.alt_hold_data)
```
Not getting anything from this script.
As your code is only 3 years old much appreciate any feedback.  Thanks

krish-iyer, 2023-09-26:
@Niko-La 

I am glad that you find my work useful. So AFAIK, if your drone support MSP and you are using something like Cleanflight then you can decide how you want arming or disarming to work(I think it's about some threshold values or something) and flash your drone. 

I think your MSP communication is working fine, you need to debug more to craft arm or disarm packets. Feel free to use Wireshark on network or USB if you want to actually see the order of values that are being sent. 

Good luck!

seanjkanderson, 2024-04-02:
@Niko-La Did you end up getting it to work? I'm trying to use a very similar route as yours. I use a Lumenier F4 AIO flight controller, which is from 2018 I believe.