alduxvm/pyMultiWii #33: Error in sendCMDreceiveATT. (tuple index out of range)

EonAndahalf, 2020-02-15:
Hello!

I'm getting an error in the terminal:

Error in sendCMDreceiveATT.
(tuple index out of range)

I'm using a Jetson nano (Ubuntu 18.04 (Python 3.6.9) ) to control a Mamba F405 FC over serial port, with the example code:

data = [1500,1500,1000,1500,1000,1000,1000,1000]
board.sendCMDreceiveATT(16,MultiWii.SET_RAW_RC,data)

Everything else is working fine except sending channel data. The weird thing is that the flight controller is receiving every channel data from my script and I can actually control the FC, except that error.

Thank you in advance!

alduxvm, 2020-02-17:
Did you activated serialRX on the GUI of the flight controller?

alduxvm, 2020-02-17:
Also, I will recommend to use the `sendCMD` function, not the send and receive...

EonAndahalf, 2020-02-19:
Hi ! 
"Did you activated serialRX on the GUI of the flight controller?"

If you mean on the Ports tab, then I have activated MSP on UART3, as that is where I connect the Jetson Nano. ( J41 GPIO pins on jetson nano to UART3 on the Flight controller ) as for the serialRX it is activated on UART1 by default. Betaflight is not allowing me to simultaneously activate MSP and serialRX on UART3. Not really sure where serialRX is supposed to be activated.

"Also, I will recommend to use the sendCMD function, not the send and receive..."

The sendCMD function requires an argument: 'data_format', and I cant seem to find what format that has to be.

EonAndahalf, 2020-02-19:
Also the "Receiver" section in the "Configurations" tab is set to: "MSP RX input (control via MSP port)"

Niko-La, 2023-09-13:
@EonAndahalf did it work?