alduxvm/pyMultiWii #34: getData type error

seanjkanderson, 2020-03-22:
When trying to run either demo in python2, I am seeing a type error on line 110 where the pack.struct is supposed to be iterated over. Here's an example output:
`angx = +0.00 	 angy = +0.00 	 heading = +0.00 	 elapsed = +0.0000 	unsupported operand type(s) for ^: 'int' and 'str'`

When debugging, it's apparent that a string is being output from iteration over the pack.struct object. Have you seen this issue before? It seems this is either something to do with the dependencies version or a a fundamental bug

alduxvm, 2020-04-20:
can you try with python3 instead?

seanjkanderson, 2020-04-26:
So it turned out that it needs to be run from the root directory of the project (i.e. demo/show-attitude.py instead of attitude.py with cwd being /demo).

I am running into an issue where I get this with some extra printout statements:
```
(108,)
(460, 210, 9)
angx = +46.00 	 angy = +21.00 	 heading = +9.00 	 elapsed = +0.0160 	()
('HEADER', '$M<\x00ll')
6
(108,)
(490, 218, 8)
angx = +49.00 	 angy = +21.80 	 heading = +8.00 	 elapsed = +0.0160 	()
('HEADER', '$M<\x00ll')
6
(108,)
Error on Main: unpack requires a string argument of length 6
```
Basically if I move the FC around quickly then this happens but if I let it sit still it will run most of the time. It seems that something is being overloaded?

In a minimal example based on your code I achieve this behavior:
```
import serial
import os
import struct
import time

ser = serial.Serial()
ser.port = "/dev/tty.usbmodem0x80000001"
ser.baudrate = 115200
ser.bytesize = serial.EIGHTBITS
ser.parity = serial.PARITY_NONE
ser.stopbits = serial.STOPBITS_ONE
ser.timeout = 0
ser.xonxoff = False
ser.rtscts = False
ser.dsrdtr = False
ser.writeTimeout = 2
ser.open()

while True:
    contents = bytes('$M<\x00ll')
    d = ser.write(contents)

    while True:
        header = ser.read().decode('utf-8')
        if header == '$':
            header = header + ser.read(2).decode('utf-8')
            break
    try:
        datalength = struct.unpack('<b', ser.read())[0]
        code = struct.unpack('<b', ser.read())
        data = ser.read(datalength)
        temp = struct.unpack('<'+'h'*int(datalength/2),data)
        print('data:', temp)
    except:
        print('dropped msg')
        ser.close()
        time.sleep(5)
        ser.open()
```
Output:
```
('data:', (-416, -163, 7))
('data:', (-419, -165, 7))
('data:', (-428, -169, 7))
('data:', (-438, -174, 7))
('data:', (-443, -177, 6))
('data:', (-452, -183, 6))
('data:', (-456, -186, 6))
('data:', (-464, -193, 6))
('data:', (-472, -199, 5))
('data:', (-476, -201, 5))
('data:', (-484, -207, 5))
('data:', (-488, -209, 5))
('data:', (-496, -214, 5))
('data:', (-504, -219, 4))
('data:', (-507, -221, 4))
('data:', (-513, -225, 4))
Traceback (most recent call last):
  File "/Users/seananderson/Documents/pyMultiWii/demo/basic.py", line 24, in <module>
    header = ser.read().decode('utf-8')
  File "/Users/seananderson/anaconda3/envs/py27/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0x8c in position 0: invalid start byte
```