alduxvm/pyMultiWii #36: cant seem to get sendCMD to work

sonnenberg16, 2020-04-18:
hi, i'm using a F4 flight controller, i set the necessary settings in betaflight, and can read the imu and attitude from the FC, however when i try using the sendCMD i get the following error: 
`Error on Main: pack expected 18 items for packing (got 26)`
this is the code i'm trynig:
```
data = [1500,1500,1000,1500,2000,1000,1000,1000] #roll/pitch/throttle/yaw/Aux1/2/3/4
for i in range(len(data)):
  binary = '{0:016b}'.format(data[i])
  data.append(int(binary[8:], 2))
  data.append(int(binary[:8], 2))
  self.sendCMD(16,MultiWii.SET_RAW_RC,data,'16B')
```

alduxvm, 2020-04-20:
is great that you're getting data! 

ok... for the commands... did you enable `SERIALRX`? also, back when I was doing this, only one control was permitted... so, do you have a remote also connect to it? try to remove it...

mmcaulif, 2021-06-28:
@sonnenberg16 Did you ever find a fix for this?

Niko-La, 2023-09-13:
@sonnenberg16 did it work?