alduxvm/pyMultiWii #38: I cannot communicate between my Raspberry pi 4 and the omnibus with MSP.

UcanYusuf, 2020-10-04:
Hello, I need to communicate between omnibus and raspberry pi 4 in my project. When I use the shared arm-disarm code, the screen show:

Board is armed now!
"In 3 seconds it will disarm..."
"Disarmed." 

However, the motors do not rotate.

Also, when I run the send data file, all values are 0 like.

I am using Betaflight.

I guess I couldn't set the MSP settings for the omnibus. Am I doing wrong in the ports part in betaflight? I am not getting errors, but the motors do not turn.

alduxvm, 2020-10-04:
ok, if you check in the readme, betaflight is not supported... maybe it works, maybe no. Please follow the instructions on the readme:
1.- in the GUI configuration for your board, activate `SERIALRX`
2.- try this script first: https://github.com/alduxvm/pyMultiWii/blob/master/demo/show-attitude.py
3.- If you get data then is a progress... if not, then the firmware is not supported and you need to try and use baseflight

UcanYusuf, 2020-10-04:
[image]
[image]
[image]
[image]

Here some outputs

UcanYusuf, 2020-10-04:
> ok, if you check in the readme, betaflight is not supported... maybe it works, maybe no. Please follow the instructions on the readme:
> 1.- in the GUI configuration for your board, activate [...]

Thank you for your reply. We got data successfully. Also, the data changes when we move the drone. This part is ok. However, the motors still do not rotate when we armed.
[image]

[image]

alduxvm, 2020-10-04:
ok, now the next step is configure properly the MSP RX input...

You send me an image before... https://user-images.githubusercontent.com/68384469/95019413-de8e4600-066d-11eb-8611-d6e6e12659ce.jpg

in that interface check that the option is selected for the MSP RX input for the usb port that you're using for reading data (USB VCP, I think...)

you also might need to remove the radiocontrol of your drone for this to work, the firmware only allows ONE method of communication, so either the USB or the radiocontrol... it cannot have both.

UcanYusuf, 2020-10-05:
> ok, now the next step is configure properly the MSP RX input...
>
> You send me an image before... https://user-images.githubusercontent.com/68384469/95019413-de8e4600-066d-11eb-8611-d6e6e12659ce.jpg [...]

Hello again, yesterday we tried a lot of things but we still have problems. Betaflight supported our controller, we can get data by using usb port. I already send picture yesterday about this. But we can't arm motors while using usb port. So we are trying to use raspberry pi4 and UART6(on flight controller). We couldn't do it "MSP RX input" configuration. Therefore, when using raspberry, we cannot receive the data like when using the usb port.

Photo 1: Rashberry-UART config
[image]

Photo2: Betaflight Status
[image]

Photo3: Betaflight Config Tab
[image]

Photo4: Betaflight Port Tab - Option 1
[image]

Photo5: Betaflight Port Tab - Option 2
[image]

Photo6: When we use this code by usb connection, we can get data from computer. But here we couldn't get data from raspberry.
[image]

Also we tried baseflight, but baseflight not supported omnibus.

UcanYusuf, 2020-10-06:
İs there a solution :( I am still trying to fix.

Niko-La, 2023-09-13:
@UcanYusuf did you figure it out?

seanjkanderson, 2024-04-02:
+1 for how to solve this @UcanYusuf