alduxvm/pyMultiWii #39: Raw RC problems

ZZ9-Plural, 2021-10-03:
I'm running multiwii on an arduino nano, and it won't let me input raw rc signals. I have success in requesting information, but raw rc commands seem not to be going through. Is there a setup procedure I have overlooked? I know that some boards need to be told what kind of rc input to receive.

alduxvm, 2021-10-04:
oh wow! its been a while since I see someone using an arduino and multiwii!! what imu are you using?? ok, in order to enable the raw rc signals, you need to enable "Serial RX", if you check the readme of this repo you will see that step prior doing raw rc signals. 

Because you are using an arduino, it means you need to find the `SERIAL_RX` constant, edit it and then compile again... as far as I remember if you enable that, you will no be able to control using a normal RC... is either one of the other... 

Let me know if it worked! Good luck!!

ZZ9-Plural, 2021-10-05:
I was unsure if that applied to the particular board I am using. I combed through the Arduino IDE to see if I could alter the SERIAL_RX value, but I have not been able to find a way. I'll keep looking, but if you have any advice, I would greatly appreciate it!

alduxvm, 2021-10-12:
thanks a lot for your input! I'm sure it will be helpful for the future!